ebelski/rust-copter #7: Running pwm-control.bat fails to produce hex file

ebelski, 2020-06-04:
@mciantyre , when running the hardware-test.bat file under the **demos** -> **pwm-control**, the compilation succeeds however the .hex file does not get produced. If the **hardware-test.sh** script is then executed directly after, the **out** folder with the associated hex file is correctly produced.

`The system cannot find the file specified.
Failed with error code 2.`

mciantyre, 2020-06-04:
Sorry I never tested the `hardware-test.bat` script!

Remove `@echo off` at the top of the script. Does that help identify where we're failing?

Couple of things we can manually check:

- show that `rmdir /S /Q out` works when `out` exists, and when `out` doesn't exist.
- show that `cargo build --release` creates a `target\thumbv7em-none-eabihf\release\pwm_control` file. If not, the `copy` command (line 14) is failing.
- try `arm-none-eabi-objcopy --help`. If it fails, we'll need to troubleshoot the toolchain installation.

ebelski, 2020-06-04:
We're failing on `rmdir /S /Q out` when out doesn't exist

ebelski, 2020-06-04:
I changed the line to read `if exist out rmdir /S /Q out || goto :error` and it looks like we're functional with or without the out directory existing.

mciantyre, 2020-06-04:
Awesome! I'll patch that upstream. Thanks for testing this.

commit dc5341c9 (2020-06-04): Changed the rmdir line to fix an error thrown when the out folder doesn't exist. Closes #7
closed 2020-06-04 by ebelski