ebelski/rust-copter #13: Error in task.py

ebelski, 2020-06-10:
```
python task.py demo pwm-control
error: Invalid character `{` in pkgid: `{crate}`
Traceback (most recent call last):
  File "task.py", line 165, in <module>
    args.func(args)
  File "task.py", line 114, in demo
    target = _cargo_build(crate, args.release)
  File "task.py", line 96, in _cargo_build
    subprocess.run(cmd, shell=True, check=True, env=env)
  File "C:\Anaconda3\lib\subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'cargo build --target thumbv7em-none-eabihf --release --package {crate}' returned non-zero exit status 101.
```

mciantyre, 2020-06-10:
Which version of Python is bundled with your Anaconda distribution?

```
python --version
```

Is it yelling about the f-string interpolation character, `{`? [f-strings](https://www.python.org/dev/peps/pep-0498/) are a Python 3.6 feature.

ebelski, 2020-06-10:
3.6.5. 

It's not passing the actual crate name, instead it's getting a literal `{crate}` in the command line

ebelski, 2020-06-10:
I think the issue lies with line 96 in task.py?

mciantyre, 2020-06-10:
This is a good catch! Just reproduced on `master`. But, I can't reproduce it on the `feature/esc-protocols` brach...?

ebelski, 2020-06-10:
There was no `f` prefix in front of the string on line 93 so it should now read `    if crate:
        cmd += f" --package {crate}"`

I'm trying that to see if it works

mciantyre, 2020-06-10:
Looks like it! Want to push a fix? Otherwise I can grab it.

I'll add some automated tests that will run

```
python task.py demo pwm-control
```

on PRs, so these kinds of bugs don't happen again.

ebelski, 2020-06-10:
I'll push an update to the repo

closed 2020-06-10 by ebelski