ebelski/rust-copter #21: Map PWM throttle percentages

ebelski, 2020-06-11:
Map PWM throttle percentages to current draw as measured by the power supply. Knowledge of the current drawn by the motor allows the force exerted by the motor/prop combo to be inferred

ebelski, 2020-06-11:
I created a spreadsheet to track this but don't know if it will upload well to github. @mciantyre if you have any suggestions for how to bring in a google sheet eloquently, that'd be much appreciated!

Otherwise, here's what I found.
[image]

I stopped at 55% throttle for 2 reasons. One, I have my power supply limited to 5.5 A current and I didn't want to saturate the supply. Two, this motor was getting pretty rowdy so I didn't want to push it just yet. I'm hoping that this half of the curve will be an ok approximation of the actual force/throttle curve.

Here's a chart of the current supplied vs. throttle percent. 
[image]

So far it looks as if a 3rd order polynomial represents the curve rather well. The only doubts that I have in the data are from the specified parameters of the motor. It calls out for our setup (2300 KV, 14.8V, 5045 prop):

- a max current load of 27.6 A (the predicted current of 100% throttle is about 24 A or a 13% difference. I could be convinced that this is within their manufacturing tolerances, however I think it is due to nonlinear torque curves as the motor reaches its maximum rated current.)
- and corresponding to the maximum current, the maximum predicted pull in grams is only 814, not 950. (I don't know why this is specified in grams, it should be specified in Newtons in my opinion or at least mN) Again this could be a tolerance issue but is more likely a nonlinear relationship. We won't know unless we measure at 100% throttle, however I don't have a way to inject 27.6 A and measure it.

[image]

ebelski, 2020-06-11:
Even a rough model will be useful in developing a state estimator control scheme. Obviously the more accurate the model the better, but I think even a 13% discrepancy is something the control loop will be robust to.

mciantyre, 2020-06-11:
LGTM!

> have any suggestions for how to bring in a google sheet [...]

- If the document can be made public, we can add a link in this issue.
- We export the sheet as a CSV, and check that into the repo. We'll treat is as read only.
- Check in the raw data, and create a Python script to auto-generate the derived data and that plot.

Any option is good with me!

ebelski, 2020-06-11:
<iframe src="https://docs.google.com/spreadsheets/d/e/2PACX-1vRwAtvcS9hT733l32kDBAb9rSdvrcgj9SYalw7AGWiev5Jk97nK2Tfy6cE2jeu-qcLKq769bl4RLJjm/pubhtml?widget=true&amp;headers=false"></iframe>

ebelski, 2020-06-12:
This issue still bothers me slightly. We're relying on the specs from this manufacturer...which I don't outright trust. We can go the more robust route and actually measure the force with a load cell

- https://www.sparkfun.com/products/14729
- https://www.sparkfun.com/products/13879

and create a true calibration of throttle percentage to force. I don't actually think this is necessary but let's keep this idea in the back of our minds should we need a more accurate quadcopter model.

closed 2020-06-12 by ebelski