ebelski/rust-copter #36: Issue with getting started script recognizing python version

ebelski, 2021-04-03:
On line 60 of the first-time-setup.sh we have a chunk of code to evaluate the python version

```bash # Check for Python installation with minimum version
# for versions. TODO check version.
if ! python -c "import sys; assert sys.version_info[0] == 3 and sys.version_info[1] > 6" &> /dev/null
then
    echo "[!] Cannot find Python interpreter at 'python', or installation is not the 3.7 min version"
    echo "[!] Make sure you have at least a Python 3.7 installation available"
else
    echo "[✓] Found Python installation with interpreter 'python' and min version 3.7"
fi
```

However when running this script, having Python 3.8 installed, it throws the errors that it cannot find the interpreter or it's not the min version. If I remove the ! from the if statement, it runs fine. Not the behavior I would expect from the statement, so clearly it's not evaluating the bool properly. I'm struggling to get any output from the statement to diagnose the issue.

mciantyre, 2021-04-04:
Does this work in the same environment?

```
python --version
```

We should see that command succeed. If it doesn't, we've failed the first check: cannot find the Python interpreter at `python`.

If it succeeds, we should see a version greater than 3.6. If it's not greater than 3.6, we've failed the second check: the `python` installation is not at least 3.7.

On my system, I have `python` pointing to a **2.7** installation, and `python3` pointing at a **3.9** installation. Is that similar to your setup? That would cause the script to fail, since we're explicitly looking for `python`, not `python3`.

We can use a [virtual environment](https://docs.python.org/3/tutorial/venv.html) to retarget the interpreter for this project (and maintain dependency isolation). After activating the virtual environment, you should find that `python` points to the interpreter used to create the environment. Use your Python 3.8 installation to create the environment.

If creating the virtual environment inside the project directory, consider naming it `pyenv`, since [it's already excluded from version control](https://github.com/ebelski/rust-copter/blob/dea5684b8a219596b16cec1fb3173d4811e71074/.gitignore#L3). Or, add the virtual environment directory to the ignore list.

> I'm struggling to get any output from the statement to diagnose the issue.

Make sure to remove the `&> /dev/null` redirection, which is designed to suppress output. Use `print` to see the machine-readable version information:

```
python -c "import sys; print(sys.version_info)"
```

ebelski, 2021-04-05:
> Does this work in the same environment?
> 
> ```
> python --version
> ```
> 
> We should see that command succeed. If it doesn't, we've failed the first check: cannot find the Python interpreter at `python`.

Yes, I get version 3.8.6.

> On my system, I have `python` pointing to a **2.7** installation, and `python3` pointing at a **3.9** installation. Is that similar to your setup? That would cause the script to fail, since we're explicitly looking for `python`, not `python3`.

I only have python3 installed on my system and in the terminal, python wasn't recognized as a command. So in a bash_aliases file I created an alias for python to point to python3. Now when I call python from terminal it points to my python3 install.

> 
> > I'm struggling to get any output from the statement to diagnose the issue.
> 
> Make sure to remove the `&> /dev/null` redirection, which is designed to suppress output. Use `print` to see the machine-readable version information:

I had removed the `&> /dev/null` and still wasn't really getting an output in terminal.

> ```
> python -c "import sys; print(sys.version_info)"
> ```

This works for me and I get python 3.8.6 version out. 

Does the first time setup script succeed for you?

ebelski, 2021-04-05:
I should add that this isn't blocking progress, but I figure we should straighten out what's going on here so that we don't get a false failure.

mciantyre, 2021-04-05:
Thanks for the info. I think the issue is the alias on your system. When you run a bash script, the alias isn't available in the script. (See [here](https://unix.stackexchange.com/questions/1496/why-doesnt-my-bash-script-recognize-aliases) for a similar issue.) This means that `python` is available in your interactive shell, but not in the `first-time-setup.sh` script. To confirm this, change all `python` to `python3` in `first-time-setup.sh`, and see if it works.

If that's the issue, there's a few different approaches we can take:

1. Change `python` to `python3` in the setup script. This makes it explicit that we need a modern Python version. Should still work for virtual environment users. This is more effort for us, since we need to update all docs and follow this practice going forward. Might not be portable; we could hit the inverse of the issue we're troubleshooting, where a user has `python` but not `python3`.
2. Use a virtual environment, which adds `python` to your environment in a way that works within Bash scripts. (This is good practice anyway, so it's not exclusive from 1.)
3. Do nothing. Treat it like the hint that it is, and expect users to know how to fix issues later.

> Does the first time setup script succeed for you?

Yup, works for me. I use a virtual environment derived from my Python 3 installation. I activate the environment before running the script, and doing other in this project.

It also works by default in these continuous integration environments, out of the box:

- [Ubuntu](https://github.com/ebelski/rust-copter/pull/37/checks?check_run_id=2269717981#step:3:277)
- [Windows](https://github.com/ebelski/rust-copter/pull/37/checks?check_run_id=2269717991#step:3:278)

The [macOS](https://github.com/ebelski/rust-copter/pull/37/checks?check_run_id=2269717964#step:3:277) interpreter is too old, but it exists as `python`. The script warns as expected.

ebelski, 2021-04-05:
I changed the statement to 
```bash if python3 -c "import sys; sys.version_info[0] == 3 and sys.version_info[1] > 6" &> /dev/null
then
```

ebelski, 2021-04-05:
Actually, I think putting an output that sums up this `python` vs. `python3` discussion may be sufficient. I'll push an update and see what you think.

ebelski, 2021-04-05:
@mciantyre , do we have dependencies downstream that imply a `python` call over a `python3` call?

mciantyre, 2021-04-05:
> I think putting an output that sums up this python vs. python3 discussion may be sufficient. I'll push an update and see what you think.

Sounds good! Feel free to change or stack changes on #37, which provides my recommendations. I'll stop rebasing that branch.

> do we have dependencies downstream that imply a python call over a python3 call?

No dependencies. It's mostly in documentation (see the README, which has an unqualified `python` call, which implicitly requires Python 3).

ebelski, 2021-04-05:
I added to the warning, pointing to a potential cause of the issue. Thanks for resolving this!

closed 2021-04-05 by ebelski