ebelski/rust-copter #39: Issue with teensy_loader_cli

ebelski, 2021-04-06:
I downloaded the teensy_loader_cli mentioned in the readme and ran into the following issue running both the cli by itself as well as the task.py.
`python task.py demo pwm-control
    Finished release [optimized] target(s) in 0.02s
Unknown MCU type "TEENSY40"
Supported MCUs are:
 - at90usb162
 - atmega32u4
 - at90usb646
 - at90usb1286
 - mkl26z64
 - mk20dx128
 - mk20dx256
 - mk66fx1m0
 - mk64fx512
 - TEENSY2
 - TEENSY2PP
 - TEENSYLC
 - TEENSY30
 - TEENSY31
 - TEENSY35
 - TEENSY36
Traceback (most recent call last):
  File "task.py", line 166, in <module>
    args.func(args)
  File "task.py", line 117, in demo
    if not args.flash or not _flash(hex_file):
  File "task.py", line 58, in _flash
    subprocess.run(cmd, shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'teensy_loader_cli --mcu=TEENSY40 -w -v target/thumbv7em-none-eabihf/release/demo-pwm-control.hex' returned non-zero exit status 1.
`
Is there a way to update the version of the cli to support 4.0?

ebelski, 2021-04-06:
I should also add that I tried specify the mcu as `--mcu=imxrt1062` as well

ebelski, 2021-04-06:
I'm closing this for now. I'm going to development on windows but after we reach a somewhat steady state, maybe v0.2 or v0.3, I'd like to run through setting this up on linux.

closed 2021-04-06 by ebelski

mciantyre, 2021-04-06:
See [their project](https://github.com/PaulStoffregen/teensy_loader_cli)'s issue tracker. I'm not sure they're releasing prebuilt binaries with Teensy 4 support right now.

It's a quick build from source. The dependencies for Ubuntu are listed under "Compiling from Source":

```
sudo apt-get install libusb-dev
git clone https://github.com/PaulStoffregen/teensy_loader_cli.git
cd teensy_loader_cli
git checkout 0a9ad5f
make
```

After that, you should find `teensy_loader_cli` in the directory. Try listing MCUs:

```
./teensy_loader_cli --list-mcus
```

mciantyre, 2021-04-06:
If there's still issues, the [graphical loader application](https://www.pjrc.com/teensy/loader.html) should work on Linux.

ebelski, 2021-04-06:
I was still running into issues with the graphical loader on Ubuntu but I think it is an issue on my part. I've reverted to using Windows and everything is working for me now.